ublox-rs/ublox #13: Consider using heapless::Vec?

jbeaurivage, 2021-06-16:
Hi,

I have just started using this crate. It is very useful, however as someone using it in a `#![no_std]` environment (Cortex-M), having to use an allocator is cumbersome. Have you considered using `heapless::Vec` instead of `std::Vec` (or letting the user decide?) Another great option could be using something like `BBQueue`.

Even though it's not strictly necessary, I think it would be a great improvement since many people using ublox devices do so on embedded systems with very limited resources.

lkolbly, 2021-06-18:
Hi! Yes, I have considered it. My current plan of action was actually to just use an internal array, probably one passed in from the user:
```
let mut backing_buf = [0; 512];
let mut parser = Parser::new(&backing_buf[..]);
```
but I was also considering a const generics approach, although that has implications for the type.

This weekend I'll dust off my prototype and see what I can do.

jbeaurivage, 2021-06-18:
Awesome, looking forward to it!

Another thing I'd like to suggest is to let the user use the raw data sent by the receiver. What I mean by that:

For instance a `PosVelTimeRef` has most of its fields as floats (lat/long are f64, velocity is f32, etc). Converting the raw u32s into floats can be very heavy on some targets, as is my case on Cortex-M0+ which doesn't have an FPU. I'm happy to use the raw u32s and convert everything to floats later on an actually capable computer :) 

Maybe I could fire up a PR for this

Dushistov, 2021-06-18:
> Another thing I'd like to suggest is to let the user use the raw data sent by the receiver.

This is already done.

> For instance a PosVelTimeRef has most of its fields as floats

No, `NavPosVelTimeRef` looks like this (you can use `cargo expand` to see type definition) :
```
pub struct NavPosVelTimeRef<'a>(&'a [u8]);
```

So is has only one filed, raw bytes. All conversation to f32/f64 happens only if you call methods to do it.

jbeaurivage, 2021-06-18:
Ah, I probably should've done my research better before commenting then! In that case, I wonder if it would be possible to add methods to return the raw individual fields (without doing the conversions)

lkolbly, 2021-06-18:
Yeah, that could absolutely be done. Unfortunately the protocol is in base-10 fixed point, so we can't do anything nifty like retrieve a [fixed](https://docs.rs/fixed/1.9.0/fixed/). All we'd need to do is add a parallel function generator alongside [get_value_lines](https://github.com/lkolbly/ublox/blob/master/ublox_derive/src/output.rs#L25), which just returns essentially just the code that `get_raw_field_code` generates (wrapped in a function, maybe `raw_<fieldname>` or something).

lkolbly, 2021-06-20:
Okay, I pushed a prototype to the master branch, I still need to document it and stuff (the CI is failing for reasons), but it should work (I compiled it for a Cortex-M0 chip I have here, but I haven't run it in a no_std environment. The tests are pretty solid though).

For heapless functionality, you can now:
```
use ublox::{Parser, FixedLinearBuffer};

let mut buffer = [0; 1024];
let buffer = FixedLinearBuffer::new(&mut buffer);
let mut parser = Parser::new(buffer);
```
and use parser much like you used to (`Parser::default()` still does a heap-based vector parser). Thinking about it more, I might add the proper impls so you can use a `heapless::Vec`, just to make handling the references easier.

For raw fields, and field which has a mapper (in particular, most lat/lon fields) now have a `_raw()` method (e.g. `fixed_alt_raw()`) which returns the raw field as whatever underlying type it is (a `i32` in the case of most of the float fields).

Unless any crazy major flaws come up, I'm hoping to push a 0.3.0 release with these changes later this week. But, feel free to give it a shot and lemme know if you have any issues.

jbeaurivage, 2021-06-21:
Awesome! Did not expect you to move that fast! I'll try it out this week and let you know if I have any feedback.

closed 2021-06-26 by jbeaurivage