ublox-rs/ublox #16: How to support repeated types?

aholtzma-am, 2022-02-09:
We are looking at adding the NavSat message, but it there doesn't seem to be a way to support repeated structs. Is there a way to do this or is it just not supported yet?

lkolbly, 2022-02-10:
Yeah, there isn't really currently a great way to support NavSat-style repeated messages.

I think the way to do it (and I can look into adding the necessary infrastructure, probably this weekend or next) would be to extend the derive macros to have an explicit notion of arrays, and then the field accessor function could take an index, something like this:
```rust
navsat_packet.num_svs(); // Number of SVs
navsat_packet.svs(5).gnssId(); // Get the gnssId of space vehicle 5
```
although, you'd want to be able to catch if the user passed an invalid index, so maybe it should look more like:
```rust
navsat_packet.svs(5).unwrap().gnss_id();
```
and I imagine a common operation is iterating over the SVs... so maybe it could just return an iterator. Something like:
```rust
for sv in navsat_packet.svs() {
    sv.gnss_id();
}
```
and, of course, you the end user would still be able to select the nth navsat by using the `nth` function:
```rust
navsat_packet.svs().nth(5).unwrap().gnss_id();
```
Does that seem like a reasonable API to you?

aholtzma-am, 2022-02-12:
This would work 100% for us.

aholtzma-am, 2022-02-14:
You might want to take a look over at: 

https://github.com/connectedcars/ublox/pull/1

They have added vec<> types to the code generator among other things. Not sure if they plan to upstream it.

CramBL, 2025-03-09:
Seems that it was resolved here https://github.com/ublox-rs/ublox/commit/02d3405fe48c2172767af89cc2ecc784a287e23c and in subsequent commits.

I'll close as resolved and anything unresolved related to `NavSat` should be described in a new issue.

closed 2025-03-09 by CramBL