ublox-rs/ublox #33: Incorrect fixed payload length for NavPosVelTime?

sorenrade, 2022-07-27:
Hi! Thanks for the great crate!

I'm working off of the u-blox 7 Protocol Specification V14 document. 

The payload length of the NavPosVelTime should be 84 bytes according to this spec. See NAV-PVT on page 158. If you are working off a different spec that no longer matches this it's probably best to note what version you support in the README. 

https://github.com/lkolbly/ublox/blob/6422f57cf98aa39e3660377de295909e40eeb499/ublox/src/ubx_packets/packets.rs#L90

lkolbly, 2022-09-11:
Hi! Thanks, sorry for my delay in responding. Yeah, most of the code is developed for the M8 protocol, although it's becoming more of an ambiguous combination.

The _goal_ is to support "all" (or at least multiple) protocols. I think this crate is going to have to be majorly refactored, I'm thinking maybe having multiple submodules for the different protocols. I had previously wanted to maintain a sort of protocol agnosticism, so you didn't have to figure out what protocol your device talked in order to use the crate. But, I think in practice people will know what device they have.

All that is to say, yes this is a huge problem.

andrei-ng, 2025-03-13:
This should now be feasible with the new features flags. Will make the changes once #94 is meged in.

commit 937814c6 (2025-04-29): introduce feature for older protocol, proto17

  - add required feature flag and refactor

Closes #33
commit a9c1eefa (2025-04-29): extract all packets into own files

 - introduce proto17 feature flag to close #33

Closes #33, #104, #43, #85
commit 611b874b (2025-04-29): extract all packets into own files

 - introduce proto17 feature flag to close #33

Closes #33, #104, #43, #85
commit d8982070 (2025-04-29): extract all packets into own files

 - introduce proto17 feature flag to close #33

Closes #33, #104, #43, #85
closed 2025-04-30 by andrei-ng