ublox-rs/ublox #52: Looking for a new maintainer

lkolbly, 2022-12-04:
Much though I would like to pretend, it is now apparent to me that I haven't the time or resources to maintain this crate like it deserves. It's been a great run and the community has really turned this into a crate to be proud of.

So, if anyone wants to take over ownership of this crate, let me know!

maximeborges, 2022-12-05:
I could help for some technical bug/features, but I don't want to be the day-to-day maintainer for this lib.
I'd happily join an org for helping though.

reitermarkus, 2022-12-05:
I'd say creating an organization (I'm guessing `ublox-rs` is still free) is the best course of action.

gwbres, 2022-12-05:
I agree.

@lkolbly what do you think about the idea of transferring to a `ublox-rs` organization?
you're obviously more than welcome onboard

sebastient, 2023-03-14:
Curious if there's been any movement on this outside this thread?

gwbres, 2023-05-18:
> Curious if there's been any movement on this outside this thread?

Not that I'm aware of. 

@lkolbly, what do you think of transfering this crate to a github orgnization?

andrei-ng, 2023-05-23:
> > Curious if there's been any movement on this outside this thread?
> 
> Not that I'm aware of.
> 
> @lkolbly, what do you think of transfering this crate to a github orgnization?

I would also be interested to contribute and a github organization would be a nice way forward.

adrian-kong, 2023-05-23:
I agree, I would be interested as well

silverjam, 2023-07-24:
Also interested in helping with maintenance

lkolbly, 2023-08-01:
Yeah, a github organization probably makes sense. I created the `ublox-rs` organization and moved this repo to it.

gwbres, 2023-08-08:
Hello @lkolbly,

> Yeah, a github organization probably makes sense. I created the ublox-rs organization and moved this repo to it

would it be possible for the people who have shown and still show interests, to join in ?

andrei-ng, 2023-09-04:
> Hello @lkolbly,
> 
> > Yeah, a github organization probably makes sense. I created the ublox-rs organization and moved this repo to it
> 
> would it be possible for the people who have shown and still show interests, to join in ?

@lkolbly , would you be so kind and willing to add some of the more active users of this repo to he organization ? It is a pity if this crate gets forked further in order to add new features & fix issues.

gwbres, 2023-09-04:
👍 

> I'm not sure if I need to add maintainers to the cargo separately, or if crates.io will pick up on the organization. We'll see.

Apparently, the process is for the owner to declare a new owner, with [cargo owner](https://doc.rust-lang.org/cargo/commands/cargo-owner.html). Apparently it is possible to declare an organization like `ublox-rs`

closed 2023-09-09 by gwbres