ublox-rs/ublox #54: UbxUnknownPacketRef

brandonros, 2023-07-28:
Hardware: Geekstory USB GPS Module Antenna Supports GPS+GLONASS Beidou Built-in Flash for Arduino Pixhawk IPEX Compatible, Better Than VK-162 https://www.amazon.com/gp/product/B07ZR8R17Q/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

```
$ uname -a
Darwin MacBook-Air.local 22.5.0 Darwin Kernel Version 22.5.0: Thu Jun  8 22:21:34 PDT 2023; root:xnu-8796.121.3~7/RELEASE_ARM64_T8112 arm64
$ cargo run -- --port /dev/cu.usbmodem11401
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `target/debug/ublox_cli --port /dev/cu.usbmodem11401`
Opened u-blox device, waiting for solutions...
Unknown(UbxUnknownPacketRef { payload: [82, 79, 77, 32, 67, 79, 82, 69, 32, 51, 46, 48, 49, 32, 40, 49, 48, 55, 56, 56, 56, 41, 0, 0, 0, 0, 0, 0, 0, 0, 48, 48, 48, 56, 48, 48, 48, 48, 0, 0, 70, 87, 86, 69, 82, 61, 83, 80, 71, 32, 51, 46, 48, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 80, 82, 79, 84, 86, 69, 82, 61, 49, 56, 46, 48, 48, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 71, 80, 83, 59, 71, 76, 79, 59, 71, 65, 76, 59, 66, 68, 83, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 83, 66, 65, 83, 59, 73, 77, 69, 83, 59, 81, 90, 83, 83, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], class: 10, msg_id: 4 })
```

brandonros, 2023-07-28:
[image]

I would guess it isn't u-blox if it didn't show up this way on System Information.... is it... a knockoff? How can I find out exactly which chip it is?

brandonros, 2023-07-28:
It sort of works with PyGPSClient which probably rules out the hardware being funky/non-complaint

[image]

I'll try to get serial TX/RX logs if that'll help us.

brandonros, 2023-07-28:
```
016: 2c32 342c 2c2c 2c2c 2c2c 2c2c 2c31 2e39  | b',24,,,,,,,,,,1.9' |
032: 322c 312e 3633 2c31 2e30 302a 3138 0d0a  | b'2,1.63,1.00*18\r\n' |

<NMEA(GNGSA, opMode=A, navMode=2, svid_01=, svid_02=, svid_03=, svid_04=, svid_05=, svid_06=, svid_07=, svid_08=, svid_09=, svid_10=, svid_11=, svid_12=, PDOP=1.92, HDOP=1.63, VDOP=1.0)>
000: 2447 4e47 5341 2c41 2c32 2c2c 2c2c 2c2c  | b'$GNGSA,A,2,,,,,,' |
016: 2c2c 2c2c 2c2c 2c31 2e39 322c 312e 3633  | b',,,,,,,1.92,1.63' |
032: 2c31 2e30 302a 3132 0d0a                 | b',1.00*12\r\n' |

<NMEA(GPGSV, numMsg=2, msgNum=1, numSV=6, svid_01=5, elv_01=28.0, az_01=56, cno_01=29, svid_02=15, elv_02=75.0, az_02=20, cno_02=12, svid_03=18, elv_03=43.0, az_03=329, cno_03=27, svid_04=23, elv_04=, az_04=, cno_04=29)>
000: 2447 5047 5356 2c32 2c31 2c30 362c 3035  | b'$GPGSV,2,1,06,05' |
016: 2c32 382c 3035 362c 3239 2c31 352c 3735  | b',28,056,29,15,75' |
032: 2c30 3230 2c31 322c 3138 2c34 332c 3332  | b',020,12,18,43,32' |
048: 392c 3237 2c32 332c 2c2c 3239 2a34 350d  | b'9,27,23,,,29*45\r' |
064: 0a                                       | b'\n' |

<NMEA(GPGSV, numMsg=2, msgNum=2, numSV=6, svid_01=24, elv_01=30.0, az_01=152, cno_01=25, svid_02=29, elv_02=, az_02=, cno_02=28)>
000: 2447 5047 5356 2c32 2c32 2c30 362c 3234  | b'$GPGSV,2,2,06,24' |
016: 2c33 302c 3135 322c 3235 2c32 392c 2c2c  | b',30,152,25,29,,,' |
032: 3238 2a34 410d 0a                        | b'28*4A\r\n' |

<NMEA(GLGSV, numMsg=1, msgNum=1, numSV=0)>
000: 2447 4c47 5356 2c31 2c31 2c30 302a 3635  | b'$GLGSV,1,1,00*65' |
016: 0d0a                                     | b'\r\n' |

<NMEA(GNGLL, lat=26.2990791667, NS=N, lon=-80.2809273333, EW=W, time=19:06:51, status=A, posMode=A)>
000: 2447 4e47 4c4c 2c32 3631 372e 3934 3437  | b'$GNGLL,2617.9447' |
016: 352c 4e2c 3038 3031 362e 3835 3536 342c  | b'5,N,08016.85564,' |
032: 572c 3139 3036 3531 2e30 302c 412c 412a  | b'W,190651.00,A,A*' |
048: 3633 0d0a                                | b'63\r\n' |

<NMEA(GNRMC, time=19:06:52, status=A, lat=26.2990801667, NS=N, lon=-80.2809288333, EW=W, spd=0.015, cog=, date=2023-07-28, mv=, mvEW=, posMode=A)>
000: 2447 4e52 4d43 2c31 3930 3635 322e 3030  | b'$GNRMC,190652.00' |
016: 2c41 2c32 3631 372e 3934 3438 312c 4e2c  | b',A,2617.94481,N,' |
032: 3038 3031 362e 3835 3537 332c 572c 302e  | b'08016.85573,W,0.' |
048: 3031 352c 2c32 3830 3732 332c 2c2c 412a  | b'015,,280723,,,A*' |
064: 3743 0d0a                                | b'7C\r\n' |

<NMEA(GNVTG, cogt=, cogtUnit=T, cogm=, cogmUnit=M, sogn=0.015, sognUnit=N, sogk=0.028, sogkUnit=K, posMode=A)>
000: 2447 4e56 5447 2c2c 542c 2c4d 2c30 2e30  | b'$GNVTG,,T,,M,0.0' |
016: 3135 2c4e 2c30 2e30 3238 2c4b 2c41 2a33  | b'15,N,0.028,K,A*3' |
032: 330d 0a                                  | b'3\r\n' |

<NMEA(GNGGA, time=19:06:52, lat=26.2990801667, NS=N, lon=-80.2809288333, EW=W, quality=1, numSV=3, HDOP=1.63, alt=2.7, altUnit=M, sep=-27.0, sepUnit=M, diffAge=, diffStation=)>
000: 2447 4e47 4741 2c31 3930 3635 322e 3030  | b'$GNGGA,190652.00' |
016: 2c32 3631 372e 3934 3438 312c 4e2c 3038  | b',2617.94481,N,08' |
032: 3031 362e 3835 3537 332c 572c 312c 3033  | b'016.85573,W,1,03' |
048: 2c31 2e36 332c 322e 372c 4d2c 2d32 372e  | b',1.63,2.7,M,-27.' |
064: 302c 4d2c 2c2a 3732 0d0a                 | b'0,M,,*72\r\n' |

<NMEA(GNGSA, opMode=A, navMode=2, svid_01=18, svid_02=5, svid_03=24, svid_04=, svid_05=, svid_06=, svid_07=, svid_08=, svid_09=, svid_10=, svid_11=, svid_12=, PDOP=1.92, HDOP=1.63, VDOP=1.0)>
000: 2447 4e47 5341 2c41 2c32 2c31 382c 3035  | b'$GNGSA,A,2,18,05' |
016: 2c32 342c 2c2c 2c2c 2c2c 2c2c 2c31 2e39  | b',24,,,,,,,,,,1.9' |
032: 322c 312e 3633 2c31 2e30 302a 3138 0d0a  | b'2,1.63,1.00*18\r\n' |

<NMEA(GNGSA, opMode=A, navMode=2, svid_01=, svid_02=, svid_03=, svid_04=, svid_05=, svid_06=, svid_07=, svid_08=, svid_09=, svid_10=, svid_11=, svid_12=, PDOP=1.92, HDOP=1.63, VDOP=1.0)>
000: 2447 4e47 5341 2c41 2c32 2c2c 2c2c 2c2c  | b'$GNGSA,A,2,,,,,,' |
016: 2c2c 2c2c 2c2c 2c31 2e39 322c 312e 3633  | b',,,,,,,1.92,1.63' |
032: 2c31 2e30 302a 3132 0d0a                 | b',1.00*12\r\n' |

<NMEA(GPGSV, numMsg=2, msgNum=1, numSV=6, svid_01=5, elv_01=28.0, az_01=56, cno_01=28, svid_02=15, elv_02=75.0, az_02=20, cno_02=12, svid_03=18, elv_03=43.0, az_03=329, cno_03=27, svid_04=23, elv_04=, az_04=, cno_04=29)>
000: 2447 5047 5356 2c32 2c31 2c30 362c 3035  | b'$GPGSV,2,1,06,05' |
016: 2c32 382c 3035 362c 3238 2c31 352c 3735  | b',28,056,28,15,75' |
032: 2c30 3230 2c31 322c 3138 2c34 332c 3332  | b',020,12,18,43,32' |
048: 392c 3237 2c32 332c 2c2c 3239 2a34 340d  | b'9,27,23,,,29*44\r' |
064: 0a                                       | b'\n' |

<NMEA(GPGSV, numMsg=2, msgNum=2, numSV=6, svid_01=24, elv_01=30.0, az_01=152, cno_01=25, svid_02=29, elv_02=, az_02=, cno_02=28)>
000: 2447 5047 5356 2c32 2c32 2c30 362c 3234  | b'$GPGSV,2,2,06,24' |
016: 2c33 302c 3135 322c 3235 2c32 392c 2c2c  | b',30,152,25,29,,,' |
[... 484 lines cut ...]
016: 2c2c 2c2c 2c2c 2c34 2e37 302c 312e 3330  | b',,,,,,,4.70,1.30' |
032: 2c34 2e35 322a 3145 0d0a                 | b',4.52*1E\r\n' |

<NMEA(GPGSV, numMsg=2, msgNum=1, numSV=6, svid_01=5, elv_01=28.0, az_01=56, cno_01=29, svid_02=15, elv_02=75.0, az_02=21, cno_02=, svid_03=18, elv_03=43.0, az_03=329, cno_03=30, svid_04=23, elv_04=31.0, az_04=294, cno_04=28)>
000: 2447 5047 5356 2c32 2c31 2c30 362c 3035  | b'$GPGSV,2,1,06,05' |
016: 2c32 382c 3035 362c 3239 2c31 352c 3735  | b',28,056,29,15,75' |
[... cut ...]
000: 2447 4c47 5356 2c31 2c31 2c30 302a 3635  | b'$GLGSV,1,1,00*65' |
016: 0d0a                                     | b'\r\n' |

<NMEA(GNGLL, lat=26.2990305, NS=N, lon=-80.2808298333, EW=W, time=19:07:02, status=A, posMode=A)>
000: 2447 4e47 4c4c 2c32 3631 372e 3934 3138  | b'$GNGLL,2617.9418' |
016: 332c 4e2c 3038 3031 362e 3834 3937 392c  | b'3,N,08016.84979,' |
032: 572c 3139 3037 3032 2e30 302c 412c 412a  | b'W,190702.00,A,A*' |
048: 3639 0d0a                                | b'69\r\n' |
```

lkolbly, 2023-07-31:
Thanks for the report! It definitely seems like this is a bug in the crate. In particular, a bug here: https://github.com/ublox-rs/ublox/blob/master/ublox/src/ubx_packets/packets.rs#L3032

That function is what decides whether the packet is "valid" or not. I'm able to reproduce the bug in a playground [here](https://play.rust-lang.org/?version=stable&mode=debug&edition=2021&gist=f6ab169aa5acdffd7cd19e8d38efbe29), which points to the issue: the `.any(|c| !is_cstr_valid(c))` will return true if any of the extensions _aren't_ valid, so we need to actually do:
```
    fn is_valid(payload: &[u8]) -> bool {
        payload.len() % 30 == 0 && !payload.chunks(30).any(|c| !is_cstr_valid(c))
    }
```
I think?

lkolbly, 2023-07-31:
I'm not sure how I never caught this. Maybe my hardware has a non-UTF8-decodable extension, and I never noticed (the original HW I developed this crate with was, in fact, somewhat less than genuine).

lkolbly, 2023-07-31:
@brandonros Can you try the `lkolbly/mon-ver-fix` branch?

Dushistov, 2023-07-31:
There is also this PR https://github.com/ublox-rs/ublox/pull/47 ,
in particular it changes `is_valid` to:

```
fn is_valid(payload: &[u8]) -> bool {
        payload.len() % 30 == 0 && payload.chunks(30).all(|c| is_cstr_valid(c))
}
```

brandonros, 2023-08-02:
Thanks for merging. Not sure if you want to trigger a full new tag bump + release. Thank you!

```
$ cargo run --bin ublox_cli -- --port /dev/cu.usbmodem1301
    Finished dev [unoptimized + debuginfo] target(s) in 0.03s
     Running `target/debug/ublox_cli --port /dev/cu.usbmodem1301`
Opened u-blox device, waiting for solutions...
SW version: ROM CORE 3.01 (107888) HW version: 00080000
MonVer { software_version: "ROM CORE 3.01 (107888)", hardware_version: "00080000", extension: MonVerExtensionIter { data: [70, 87, 86, 69, 82, 61, 83, 80, 71, 32, 51, 46, 48, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 80, 82, 79, 84, 86, 69, 82, 61, 49, 56, 46, 48, 48, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 71, 80, 83, 59, 71, 76, 79, 59, 71, 65, 76, 59, 66, 68, 83, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 83, 66, 65, 83, 59, 73, 77, 69, 83, 59, 81, 90, 83, 83, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], offset: 0 } }
```

closed 2023-08-02 by brandonros