ublox-rs/ublox #55: NavPosVelTime packet not being received?

brandonros, 2023-08-02:
```
MacBook-Air:ublox brandon 2023-08-02 13:05:13 $ cargo run --bin ublox_cli -- --port /dev/cu.usbmodem1301
   Compiling ublox_cli v0.1.0 (/Users/brandon/Desktop/ublox/ublox_cli)
    Finished dev [unoptimized + debuginfo] target(s) in 0.29s
     Running `target/debug/ublox_cli --port /dev/cu.usbmodem1301`
Opened u-blox device, waiting for solutions...
SW version: ROM CORE 3.01 (107888) HW version: 00080000
MonVer { software_version: "ROM CORE 3.01 (107888)", hardware_version: "00080000", extension: MonVerExtensionIter { data: [70, 87, 86, 69, 82, 61, 83, 80, 71, 32, 51, 46, 48, 49, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 80, 82, 79, 84, 86, 69, 82, 61, 49, 56, 46, 48, 48, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 71, 80, 83, 59, 71, 76, 79, 59, 71, 65, 76, 59, 66, 68, 83, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 83, 66, 65, 83, 59, 73, 77, 69, 83, 59, 81, 90, 83, 83, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], offset: 0 } }
```

how long would you expect to have to wait before seeing `NavPosVelTime` come across? would it come across if there was no fix/it was in the process of getting a fix?

brandonros, 2023-08-02:
I have some trace logs but I'd rather not put them public because... well... they're GPS coordinates :P (the device does get a fix)

I don't think we can reliably just send the serial received  bytes into a parser that expects to be `0xB5` delimited?

It's just spitting out newline delimited (`0x0A`) NMEA: http://aprs.gids.nl/nmea/

I don't even see the need to send the device I have `CfgPrtUartBuilder` or `CfgMsgAllPortsBuilder::set_rate_for::<NavPosVelTime>([0, 1, 0, 0, 0, 0])` packets. Apparently you just plug it straight in and it just spits NMEA? Sorry if this was known.

gwbres, 2023-08-03:
Hello @brandonros,

> Apparently you just plug it straight in and it just spits NMEA? Sorry if this was known.

it all depends on which protocol you enabled or not, you need to use the protocol configuration frame to set it up.  
I can't remember whether I had to do it in my fork, or the current state of this lib allows it.  
This lib has not really been updated for 1 or 2 years now, it's becoming a maintenance issue

stevenlovegrove, 2023-08-22:
Great library! For anyone else who reads this thread, make sure you check the readme at head of this repo (and not the outdated one on crates.io) which includes the CfgValSet packet builder which makes configuration for devices like the ublox f9p nice and easy. Here is a modified gist of the cli example but enabling ubx output over usb for example. This is what I needed to get anything other than the version out of the example.

https://gist.github.com/stevenlovegrove/4f46f9f54414d2be68f04309eb685721

andrei-ng, 2023-11-09:
@brandonros, can this issue be closed ? Were you able to succesfully receive NAV-PVT from your device?

closed 2023-11-09 by brandonros