ublox-rs/ublox #68: Release v0.4.3 needs a new release of ublox_derive

mattico, 2023-09-26:
Currently, trying to use `ublox = "0.4.3"` from crates.io fails with errors like:

```
error: Not supported type, expect one of "u8, i8, u16, i16, u32, i32"
    --> C:\Users\matth\.cargo\registry\src\index.crates.io-6f17d22bba15001f\ublox-0.4.3\src\ubx_packets\packets.rs:1709:15
     |
1709 |     cfg_data: &'a [CfgVal],
     |   
error: expected one of: `map_type`, `scale`, `alias`, `may_fail`, `from`, `is_valid`, `get_as_ref`, `into`
    --> C:\Users\matth\.cargo\registry\src\index.crates.io-6f17d22bba15001f\ublox-0.4.3\src\ubx_packets\packets.rs:3069:9
     |
3069 |         size_fn = data_len,
     |    
error[E0412]: cannot find type `EsfMeas` in this scope
    --> C:\Users\matth\.cargo\registry\src\index.crates.io-6f17d22bba15001f\ublox-0.4.3\src\ubx_packets\packets.rs:3082:6
     |
3082 | impl EsfMeas {
     |      ^^^^^^^ not found in this scope
     |
help: there is an enum variant `crate::PacketRef::EsfMeas`; try using the variant's enum
     |
3082 | impl crate::PacketRef {
     |      
```

This is because v0.4.3 requires the changes which have been made to ublox_derive since its v0.0.4 release. Overriding the ublox_derive dependency to use the new code fixes the errors, e.g.

```toml
[patch.crates-io]
ublox_derive = { git = "ssh://git@github.com/ublox-rs/ublox.git", tag = "ublox-v0.4.3" }
```

I also noticed that the `ublox_derive-v0.0.4` tag points to the same commit as `ublox-v0.4.3`, rather than its release commit which I think is the tag `v0.4.0`.

One other minor thing I noticed is that 0.4.3 on crates.io includes the changes from #67 which is after the `ublox-v0.4.3` tag, so the `ublox-v0.4.3` tag could be changed to [8f3e5e2](https://github.com/ublox-rs/ublox/commit/8f3e5e2686a16b299b8e971f6c19256434fedbc0). I checked this by using `cargo vendor` and diff against a git checkout.

I see that this crate was moved to an org recently. I'd be happy to help out maintaining this crate if you need an extra set of hands!

andrei-ng, 2023-09-26:
I think your finding is great stuff! If not mistaken, this is related to why publishing currently fails , as per this draft PR https://github.com/ublox-rs/ublox/pull/66 . 

I think @gwbres is currently the maintainer. It would be nice if he could add you to the org.

gwbres, 2023-09-27:
Hello @mattico 

I'm happy to add you to the repo, but I have no means to add you to the organization.

I'll see if I can publish a new _derive_ release soon

closed 2023-09-28 by gwbres