ublox-rs/ublox #78: How to get length of parsed packet?

blowback, 2024-05-11:
Given a `PacketRef` obtained from a parser iterator, how do you get the packet's original encoded length? There's a `class_and_msg_id()` method on PacketRef, but I don't see anything similar for length (bytes at offsets 4 and 5).

gwbres, 2024-05-11:
@blowback,

We have no means to expose such information in the current state, when dealing with complex packets whose size may vary.
With simplistic packets, you have UbxPacketMeta::FIXED_PAYLOAD_LEN.
To answer that requirement, `pack_len` [here should be saved in the packet being constructed](https://docs.rs/ublox/latest/src/ublox/parser.rs.html#496)

andrei-ng, 2025-09-28:
@blowback , every `PacketRef` has now an associated method ` packet.payload_len() ` to get the payload lenght.

commit e5495747 (2025-09-28): add packet-ref method to get packet payload length

Closes #78
commit c4b710c3 (2025-09-28): add packet-ref method to get packet payload length

Closes #78
commit 84c013ed (2025-09-28): add packet-ref method to get packet payload length

Closes #78