ublox-rs/ublox #84: Missing examples for CfgValSetBuilder

yorickbrunet, 2025-03-04:
Hello,

I'm trying to use the crate but run on different issues due to missing documentation / examples.

I'd like to configure my ublox device to output only UBX messages. So I started with:

```rust
let cfg_data: [ublox::cfg_val::CfgVal; 4] = [
            ublox::cfg_val::CfgVal::Uart1InProtUbx(true),
            ublox::cfg_val::CfgVal::Uart1InProtNmea(false),
            ublox::cfg_val::CfgVal::Uart1OutProtUbx(true),
            ublox::cfg_val::CfgVal::Uart1OutProtNmea(false),
        ];
let cfg = ublox::CfgValSetBuilder {
    version: 0,
    layers: ublox::CfgLayer::RAM,
    reserved1: 0,
    cfg_data: &cfg_data,
};
```

but how to convert `cfg` into an array on `u8` to send it over serial ?

Thanks.

gwbres, 2025-03-07:
Hello,

(I'm just getting back to this library after years, so I'm learning as well)

I've never used the `CfgVal` structures personnaly, I've only configured a few things at a time, for each individual sessions.
[For example like this](https://github.com/rtk-rs/ubx2rinex/blob/main/src/device.rs#L146). 

To turn everything off but UBX you would use  `CfgPrtUartBuilder` in the example of the UART port, and only assert `OutProtoMask:UBLOX`. So right now I can't answer anything about CfgVal, I don't really grasp its purpose

gwbres, 2025-03-16:
@andrei-ng, is that part of the API actually functional ?

It looks to me `CfgVal` is the only option we have currently to reprogram the GNSS settings,  `CfgGnss` does not exist and I can't have it compile, due to macro issues (`ubx(packet_send)` + `Iter`). Hence the importance to be able to use this side of the API

yorickbrunet, 2025-03-16:
Hi,
Thanks for your answer.
I took actually some time to play with the API and `.extend_to` is the way to go.
Working example:

```rs
const BUF_OUT_SIZE: usize = 256;
let mut bufout: heapless::Vec<u8, BUF_OUT_SIZE> = heapless::Vec::new();
let cfg_data: [ublox::cfg_val::CfgVal; 5] = [
    ublox::cfg_val::CfgVal::Uart1InProtUbx(true),
    ublox::cfg_val::CfgVal::Uart1InProtNmea(false),
    ublox::cfg_val::CfgVal::Uart1OutProtUbx(true),
    ublox::cfg_val::CfgVal::Uart1OutProtNmea(false),
    ublox::cfg_val::CfgVal::MsgoutUbxNavPvtUart1(1),
];
ublox::CfgValSetBuilder {
    version: 0,
    layers: ublox::CfgLayer::RAM,
    reserved1: 0,
    cfg_data: &cfg_data,
}
.extend_to(&mut bufout);
```

I'd be very happy if the crate and its doc would evolve as I plan to use some ublox chips with Rust in the future :)

closed 2025-03-16 by yorickbrunet

gwbres, 2025-03-17:
Before your post (I havent tried again yet) I was getting a Nack on my side after that request.
A "simple" `.extend_to(&mut [u8])` does not compile either

> I'd be very happy if the crate and its doc would evolve as I plan to use some ublox chips with Rust in the future :)

we will publish a new release this week, the README will be integrated as well

andrei-ng, 2025-03-17:
> [@andrei-ng](https://github.com/andrei-ng), is that part of the API actually functional ?
> 
> It looks to me `CfgVal` is the only option we have currently to reprogram the GNSS settings, `CfgGnss` does not exist and I can't have it compile, due to macro issues (`ubx(packet_send)` + `Iter`). Hence the importance to be able to use this side of the API
> 
> The readme proposes:
> 
> let cfg : Vec<u8> = ublox::CfgValSetBuilder {
>     version: 0,
>     layers: ublox::CfgLayer::RAM,
>     reserved1: 0,
>     cfg_data: &cfg_data,
> }.into_packet_vec()
> Which no longer exists. That makes me realize that the README is not tested and verified nor included.
> 
> The best I have is
> 
> let cfg = ublox::CfgValSetBuilder {
>     version: 0,
>     layers: ublox::CfgLayer::RAM,
>     reserved1: 0,
>     cfg_data: &cfg_data,
> };
> 
> cfg.extend_to(&mut buf);
> Which I doubt is what is intended and will work

Hi @gwbres , yes as @yorickbrunet mentions it works. Indeed, I only used it once in the past and I did it as @yorickbrunet did. 
`CfgGnss` is deprectated from version 23.01 onwards. And as I saw from the F9R protocol specification, a lot of messages in newer protocol series are deprecated in favor of `CfgValSet` and `CfgValGet`. 

We need to fix the documentation (maybe we should consider an mdBook as well) and we need to integrate #56 into `0.6.0` release as that alreay has the work for `CfgValGet`

gwbres, 2025-03-21:
@andrei-ng thank you for the clarifications.

In the long run, it will be interesting to provide support of `CfgGnss` anyway.

@yorickbrunet 

I confirm that compilation passes, but for example `[cfg_data::CfgVal::GpsL1Ena(false)]`  has no effect on my side

gwbres, 2025-03-21:
that's because I'm using M8 series and this is only supported starting from F9 series.
So I'll be one of those interested in unlocking CfgGnss