ublox-rs/ublox #85: Improve repo organization

gwbres, 2025-03-07:
Hello, @andrei-ng, @maximeborges,

in the `repo_organization` branch, I'm trying to improve the overall organization of this lib, the `packet` module has grown impossible to maintain, with up to 4K lines of code.

The problem being all the macros involved in this crate, which personnally I don't get anything from.

The PacketRef definition is being augmented with `ubx_packet_recv` which complains that all its members should be public.
All I'm trying to do, is split the 4k lines of code into separate modules, where all structures are public and inner values are also public, so this does not stand to me. Do you gus have any ideas ?

CramBL, 2025-03-09:
I have not looked at refactoring the whole module but I refactored `PacketRef` out into a separate module here: https://github.com/andrei-ng/u-blox.rs/pull/48, maybe my comments help you a bit? Also [cargo expand](https://github.com/dtolnay/cargo-expand) is invaluable here.

Are you now maintaining this crate? I've been focusing on the fork here: https://github.com/andrei-ng/u-blox.rs so I'm curious what the strategy is right now?

I'll gladly continue refactoring `packet.rs` but I need to know where to send patches to.

gwbres, 2025-03-09:
Hello @CramBL 👍 

> maybe my comments help you a bit? Also [cargo expand](https://github.com/dtolnay/cargo-expand) is invaluable here

Yeah totally, I actually discovered it yesterday while doing this

> Are you now maintaining this crate?

We've set up this community to continue maintain this library. I've personnally been busy on other topics since setup time, but now I have concluded all of them and will start gradually back to using UBX more often.
I was not aware @andrei-ng had a fork, I even thought he was part of the team right here.

IMO, we should have what you guys improved on your side, integrated here. I see that it is about 50 commits.
Well, I'll start off with two things, I'll add you guys as collaborator of the project, and I'll review a part of the modifications tonight

CramBL, 2025-03-09:
Your plan sounds good, I expect @andrei-ng will agree but let's hear from him so we can focus the effort and avoid duplicate work.

I'm very new to ublox, just starting to get into it, but on the language side I can definitely help with reviews and refactors. I can also help out with CI. e.g. if we want code coverage like this [PR with codecov summary](https://github.com/CramBL/dependabot-changelog-writer/pull/58) and/or caching etc. I can set it up.

gwbres, 2025-03-09:
>  let's hear from him so we can focus the effort and avoid duplicate work

👍  sounds good.

andrei-ng, 2025-03-09:
Hi,

@gwbres, sorry for my delayed response. I was stuck on `packets.rs` split  issue for a while, but @CramBL has the knowledge to do it as he mentions above. 

I actually don't want to maintain forks, unless I absolutely need to. I needed quite a few of the additions I made for my work and it looked you guys were busy with merging in the current PRs in this repo. Hence my fork and the changes.

However, based on this discussion and the things that have happend here today, I think we are on a good track to breathe some new life into this repo.

Thank you for adding me as maintainer!  Now I am truly part of the team 😄 

@CramBL seems a godsent 😁 , so I propose he makes the same changes he made in my fork here related to the `packet.rs` split.

I am willing to do the grunt work and pull all the packet additions I have in my fork here as separate PRs for you guys to approve. Most of my additions are related to ESF (extended sensor fusion) for robotics/automotive. I also added a bunch of new examples, including a simple TUI bases on `ratatui` because I simply love the terminal and dislike running `u-center` with `wine` (I have a feeling that we might have more users of this crate who would love this) .

Lastly, IMO we should start merging in as many of the outstanding PRs as possible. I was reviewing all of them in my fork and all have important contributions.

gwbres, 2025-03-09:
>  so I propose he makes the same changes he made in my fork here related to the packet.rs split.

hey, this should not involve any developpement, we're simply talking about merging the commits here, and trying to preserve your history.

We have several options, I would say the best one is to split into 2, 3, 4 PR tops, per major topics that you consider. No need to worry and  provide each commit individually. To create that, simply go to your fork, checkout the repo at one of those important stages (starting from oldest) and checkout into a new branch. And simply open a PR from github for that branch.

CramBL, 2025-03-09:
> I am willing to do the grunt work and pull all the packet additions I have in my fork here as separate PRs for you guys to approve. Most of my additions are related to ESF (extended sensor fusion) [...]

I ran the TUI example on a ZED-F9P last Friday and I loved it! Very cool example.

andrei-ng, 2025-03-10:
> We have several options, I would say the best one is to split into 2, 3, 4 PR tops, per major topics that you consider. No need to worry and provide each commit individually. To create that, simply [...]

Agreed! Sounds good. Will get down to it starting tomorrow.

gwbres, 2025-03-12:
We have zero means to publish a library update as of right now, so it is the priority to unlock this.
Eventually, we should have a git tag CI release job, it is the most convenient for future evolution

CramBL, 2025-03-12:
> We have zero means to publish a library update as of right now, so it is the priority to unlock this. Eventually, we should have a git tag CI release job, it is the most convenient for future evolution

Do you have anyone you can contact or are we stuck here?

lkolbly, 2025-03-12:
I have the permissions, I just gave them to gwbres in https://github.com/ublox-rs/ublox/issues/88 so y'all should be set. I appreciate y'alls taking over the crates!

andrei-ng, 2025-04-30:
Closing this issue. Crate structure has been improved in https://github.com/ublox-rs/ublox/pull/129 and https://github.com/ublox-rs/ublox/pull/130

closed 2025-04-30 by andrei-ng