ublox-rs/ublox #88: Ublox publishing permission

gwbres, 2025-03-12:
@lkolbly,

you are the only person with publishing permission to the crates I/O portal, until then the libraries will remain stuck.
You will have to help us.

We have several options

- upgrade my role to "owner" in the cargo-owner team you created
- declare me on crates I/O as owner of the ublox_derive and ublox libraries

until then, we have no means to update the libraries.
If I have permission, I can then generate tokens and generate a publish job for example, this would facilitate everybody's work

lkolbly, 2025-03-12:
Ah, sorry about that - I just made you owner of ublox-rs, and sent invites to directly own the crates also. Let me know if that doesn't work, thanks!

gwbres, 2025-03-12:
@lkolbly, thank you so much for the rapid answer,

We should be all set but we can only confirm after next release. I'm confident this will work.
Do not hesitate to take part in the future of this lib ;)

closed 2025-03-12 by gwbres