ublox-rs/ublox #104: Split packets per module

andrei-ng, 2025-04-15:
Split [packets.rs](https://github.com/ublox-rs/ublox/blob/master/ublox/src/ubx_packets/packets.rs) per protocol specification.

Once PRs https://github.com/ublox-rs/ublox/pull/56 and https://github.com/ublox-rs/ublox/pull/103 are closed. 

Closes #43

andrei-ng, 2025-04-25:
@CramBL, @gwbres what is your preference for approaching this?
Should we split [packets.rs](https://github.com/ublox-rs/ublox/blob/master/ublox/src/ubx_packets/packets.rs) 
1. into messages per protocol
2. into individual messages

Option 1: doesn't really resolve the size of the file as the majority of UBX messages are still valid for UBX Protocol 23.
Option 2: is more attractive to me, IMO it will be easier to maintain in terms of different protocol versions but we will end up with a large number of files (but this could be an advantage, we would have `ubx-na-pvt.rs`, `ubx-nav-att.rs`, `ubx-mon-ver.rs` etc. which would nicely align with the spec)

Any thoughts?

CramBL, 2025-04-25:
I prefer option 2.

I like many files more than some people I think.

The best thing about Java is that you are only allowed one class per file and the class name has to match the file name, that's great for navigating a project. I try to do something similar within reason for Rust projects.

Although I would definitely prefer snake_case rather than kebab-case for file names. Cargo does the same, e.g. if you run 'cargo new my-app' you get a new project under 'my_app'. And on crates.io it also translates hyphen to underscore. Also, underscore plays better with filename completion in a terminal.

andrei-ng, 2025-04-25:
Sounds good. I see @gwbres gave is thumbs up too, so Option 2 it is.

PR #129 (2025-04-30): Separate packets in modules
Closes  #104, #43, #85 and #33 (I've added a new feature flag for older protocol versions)
closed 2025-04-30 by andrei-ng