ublox-rs/ublox #136: Access UBX-MGA-XXX information

gwbres, 2025-05-09:
Hello @CramBL, @andrei-ng,

I don't understand how we can access `UBX-MGA-XXX` information, for instance I'd be interested in receiving the keplerian parameters using `MGA-GPS-EPH`  that I introduced years ago (never got to use it...), it is said to be an "input frame", while I would expect a Poll/Request frame. 

And indeed, if I request or try to set up the polling of this frame, it does not stream anything out. 

Is there something I'm missing here? The `AID` messages are said to be deprecated and "replaced by MGA" messages. AID frames were marked as polling and request frames. So I'm assuming MGA should work the same.

gwbres, 2025-05-09:
hum, not quite what I was expecting.

So indeed `UBX-MGA` seems to be an input only frame, but I'll wait for you guys to confirm.

I have two options
 - use `RXM-SBFR` frame that we already have, and decode the raw subframes.
I've just tested and that seems to work
 - Introduce AID-EPH and do the same thing.

If my new understand is true, I don't understand why they say MGA replaces AID frames, that is partially incorrect

gwbres, 2025-05-09:
> hum, not quite what I was expecting.
>
> So indeed `UBX-MGA` seems to be an input only frame, but I'll wait for you guys to confirm.
>
> I have two options
>
> * use `RXM-SBFR` frame that we already have, [...]

I've started a branch (`sfrbx-protos`) where I explore simple options to propose GPS words interpretation

gwbres, 2025-05-09:
I'm not a super fan of having this integrated to this library. Assuming we had an external interprator, all we have to do to make the data streamed from the ublox compliant with the GPS protocol, is to split the u32 iterator into u8, and swap the data bits.

What do you guys think is best ? 

- propose basic interpration like I started doing, directly in ublox
- or move this to another library, and propose a small bridge, like we could for RTCM for example

CramBL, 2025-05-09:
I'm trying to follow but I don't really understand it, @andrei-ng would be able to answer it I assume.

So this is a device that talks UBX? What device is it? If it talks UBX then I think it would be great to support it.

What is the general problem? Is it that the current UBX parser makes assumption about the protocol that breaks down for this UBX-MGA-XXX messages?

gwbres, 2025-05-09:
Hello,

no I'm trying to access the GPS ephemerides from any Ublox using the protocol. It would the orbital equivalent to the RXM-RAWX frame. 

I thought that was the purpose MGA frames served, but MGA frames apparently only work from the host to the Ublox (upload) not download. The only way to obtain the information I'm looking for, is the UBX-SFRBX frame, which exposes the raw GNSS protocol (constellation dependent). We support the UBX-SFRBX as of today, and expose the bytes as is. I've only studied the GPS case this afternon. In case of GPS (have not looked at others yet), the ublox protocol breaks the GPS protocol. Not exactly how I said a few hours ago, but as explained in the ublox forums : https://portal.u-blox.com/s/question/0D5Oj00000cj8VDKAY/invaid-output-on-ubxrxmsfrbx. The parity bits are truncated and lost, the bytes are shifted bytewise.

So I am trying to figure out ways to help the user access the decoded ephemeris here. 
From what I'm starting to understand, we either leave it as it is today (so they do it themselves) or we will have to do the decoding ourselves, because a regular GPS parser will not work correctly from what ublox devices stream

CramBL, 2025-05-10:
Thanks for the explanation, I'm in favor of integrating that functionality in the library. It might not fit the existing functionality, but that's essentially a bug in the ubx protocol it sounds like.

gwbres, 2025-05-10:
Hello,

> Thanks for the explanation, I'm in favor of integrating that functionality in the library. It might not fit the existing functionality, but that's essentially a bug in the ubx protocol it sounds like.

I would not call it a bug, its just how their firmware behaves. It obviously does the decoding on its side, but strips some of the bits

andrei-ng, 2025-05-10:
Sorry for my late reply. It is a busy period. 

I don't have much experience with ephemeris data, so unfortunately can't help here, I am like a fish out of water. 

Indeed, according to M8 protocol specification the MGA messages are only for sending them to the receiver. Also according to this thread https://portal.u-blox.com/s/question/0D52p0000EANJiMCQX/using-one-receiver-to-aid-another `UBX-SFRBX` is the only way.

>  So I am trying to figure out ways to help the user access the decoded ephemeris here.
From what I'm starting to understand, we either leave it as it is today (so they do it themselves) or we will have to do the decoding ourselves, because a regular GPS parser will not work correctly from what ublox devices stream

@gwbres , I agree with @CramBL , we should support the decoding in our crate. This would be a nice and useful feature of our crate.

gwbres, 2025-05-10:
Hello, yes I concur the newer frames are for so called "hot-start" (faster first fix). 
Assuming you have an ephemeris database (usually, a secondary - already started receiver), you would feed those frames to the first receiver, to make it start faster. Will be nice to make sure we support that as well.

I have just concluded GPS markers and subframe #2 in the draft branch. So the principle is there. My plan is to support GPS + Gal for starters. So not all of it, because that would be too much.

I also saw that people have made that on their side, for example this person did it for BeiDou: https://github.com/knrl/GNSS-BeiDou-Message-Decoding.

Indeed, it would be super cool if this library could leviate a little of the complexity for the end user once again. (so basically, people interested in manual navigation, like me)

gwbres, 2025-05-18:
Conclusion:

- UBX-MGA works both ways and is the modern replacement of UBX-AID (GPS only, deprecated).
- UBX-SFRBX provides raw orbital information, which is what I'm interested in and we're still lacking a little bit of support on that side

closed 2025-05-18 by gwbres